DerOetzi/solaredge2mqtt #45: Handle inverter without a battery

trickv, 2024-03-18:
My inverter doesn't have a battery, but the code is behaving as if there are two batteries - each of which has no data to report.  So when trying to instantiate a SunSpecBattery in _map_batteries, it throws InvalidModbusData which then discards all data from the inverter.  Debug output:
```
2024-03-18 13:10:12.857 | DEBUG    | solaredge2mqtt.service.modbus:_map_inverter:72 - Inverter raw:
{
    "c_id": "SunS",
    "c_did": 1,
    "c_length": 65,
    "c_manufacturer": "SolarEdge",
    "c_model": "SE5000H-US000BNU4",
    "c_version": "0004.0013.0040",
    "c_serialnumber": "7300251E",
    "c_deviceaddress": 1,
    "c_sunspec_did": 102,
    "c_sunspec_length": 50,
    "current": 357,
    "l1_current": 357,
    "l2_current": 0,
    "l3_current": 0,
    "current_scale": -2,
    "l1_voltage": 2424,
    "l2_voltage": 0,
    "l3_voltage": 0,
    "l1n_voltage": 0,
    "l2n_voltage": 0,
    "l3n_voltage": 0,
    "voltage_scale": -1,
    "power_ac": 8514,
    "power_ac_scale": -1,
    "frequency": 59978,
    "frequency_scale": -3,
    "power_apparent": 8668,
    "power_apparent_scale": -1,
    "power_reactive": 16281,
    "power_reactive_scale": -2,
    "power_factor": 9822,
    "power_factor_scale": -2,
    "energy_total": 19658620,
    "energy_total_scale": 0,
    "current_dc": 23368,
    "current_dc_scale": -4,
    "voltage_dc": 3699,
    "voltage_dc_scale": -1,
    "power_dc": 8644,
    "power_dc_scale": -1,
    "temperature": 3783,
    "temperature_scale": -2,
    "status": 4,
    "vendor_status": 0
}
2024-03-18 13:10:12.858 | DEBUG    | solaredge2mqtt.service.modbus:_map_inverter:78 - info=SunSpecInfo(manufacturer='SolarEdge', model='SE5000H-US000BNU4', option=None, sunspec_type='Split Phase Inverter', version='0004.0013.0040', serialnumber='7300251E') ac=SunSpecAC(current=SunSpecACCurrent(actual=3.57, l1=3.57, l2=0.0, l3=0.0), voltage=SunSpecACVoltage(l1=242.4, l2=0.0, l3=0.0, l1n=0.0, l2n=0.0, l3n=0.0), power=SunSpecACPower(actual=851.4, reactive=162.81, apparent=866.8, factor=98.22), frequency=59.98) dc=SunSpecDC(current=2.34, voltage=369.9, power=864.4) energytotal=19658620.0 status='Producing'
2024-03-18 13:10:12.859 | INFO     | solaredge2mqtt.service.modbus:_map_inverter:79 - Inverter (SolarEdge SE5000H-US000BNU4 7300251E): Producing, AC 851.4 W, DC 864.4 W, 19658.62 kWh
2024-03-18 13:10:12.861 | DEBUG    | solaredge2mqtt.service.modbus:_map_batteries:123 - Battery Battery1 raw:
{}
2024-03-18 13:10:12.861 | WARNING  | solaredge2mqtt.service:run_loop:181 - Invalid modbus data, skipping this loop
```

Patch forthcoming.

DerOetzi, 2024-03-20:
Please check with v1.0.2 and close this issue if it works

trickv, 2024-03-21:
Works in v1.0.2 :)

closed 2024-03-21 by trickv