DerOetzi/solaredge2mqtt #49: Docker: release image versioneer can't detect version

trickv, 2024-03-24:
I noticed this when running it in the ghcr images - that the version string printed on startup doesn't work. Tested against `ghcr.io/deroetzi/solaredge2mqtt:v1.0.3`

Reading into it, the version string is handled by versioneer, which can only get state from the .git directory or from the release package.

I had a look at the container build scripts but it's over my head.  One idea I had was to extract the version string from GitHub in build-project.yml and define VERSIONEER_OVERRIDE environment variable, but I'd need to learn a lot about GitHub build actions to play with it.

On startup it's this line:
```
2024-03-24 15:55:40.729 | INFO     | solaredge2mqtt.service:main_loop:90 - Version: 0+unknown
```

DerOetzi, 2024-03-25:
Is fixed

closed 2024-03-25 by DerOetzi

trickv, 2024-03-26:
Success - confirmed:
```
2024-03-25 21:54:52.177 | INFO     | solaredge2mqtt.service:main_loop:90 - Version: 1.0.5
```