DerOetzi/solaredge2mqtt #93: Docker läßt sich nicht starten

skipperkm, 2024-07-17:
Hi,

leider bekomme ich den Docker nicht zum laufen. Das Skript auf der Konsole läuft. Also funktioniert die .env. Ich habe die Vermutung das der Docker die .env nicht findet.

Was kann ich tun ?

DerOetzi, 2024-07-17:
Hallo,

grundsätzlich gibt es mehrere Lösungen:

1. Das .env File als Volume mounten

`docker run -v .env:.env`

2. Verwende docker-compose mit der im Verzeichnis liegenden docker-compose.yaml aus dem Projektverzeichnis oder falls du den vollen Stack mit InfluxDB und Grafana haben willst aus dem examples Verzeichnis

3. Source die .env File in der Konsole und übergebe die einzelnen benötigten Environment variablen einzeln. (Denke zu umständlich)

skipperkm, 2024-07-17:
Danke für die schnelle Antwort, ich wußte nicht wohin die .env gemountet werden muß. Ich probiere es morgen aus.

Werden mit dem Script eigentlich auch die Daten der Optimierer abgerufen?

Mit freundlichen Grüßen

DerOetzi, 2024-07-18:
Gerne berichten ob es so geklappt hat, meine persönliche Empfehlung wie gesagt wäre docker-compose zu nehmen.

Ja die Werte werden ausgelesen. Wenn du die Infos für den Bereich Monitoring konfiguriert, dann werden alle 10 Minuten die Werte ausgelesen und an MQTT gepublished.

skipperkm, 2024-07-18:
Hi,
Hat ist alles am laufen, prima. (docker-compose) Ich habe die Daten auch in meine vorhandene lnflux/Grafana Umgebung eingebunden. Hat auch nach etwas abwarten funktioniert.
Die Moduldaten habe ich gefunden. Ich meinte aber die aktuellen Leistungsdaten der Optimierer. Die können auch über das Monitoring abgerufen werden. Habe das über eine NodeRed Intragration die die Daten in die influxdb pustet.
Mir wäre es aber lieber wenn das auch über Mqtt laufen würde. Bekomme ich leider nicht hin.
Wäre das nicht eine nette Erweiterung für das Script?

Mit freundlichen Grüßen

DerOetzi, 2024-07-18:
Schön das es geklappt hat.

Mir ist neu, dass man auch die aktuellen Leistungsdaten abfragen kann, kannte bisher nur die Durchschnittswerte für eine Viertelstunde und da sehe ich nicht wirklich den Mehrwert im Vergleich zu dem in diesem Zeitraum erzeugten Energie. Aber erstell doch bitte gerne ein neues Issue als feature request. Und pack mir soviele Infos wie möglich rein gerne auch deinen node-red flow (ohne Zugangsdaten!) dann sollte es nicht all zu schwierig sein den MonitoringSite Service entsprechend zu erweitern.

skipperkm, 2024-07-18:
Mach ich

skipperkm, 2024-07-18:
Hi,

soweit ich das sehen kann sind das auch die Viertelsundenwerte.  Du hast Recht, wenn du nur module gleicher Leistung überwachen möchtest, dann reicht der Energiewert aus.
Der Punkt ist folgender wenn ich Module unterschiedlicher Leistung (kWp) vergleichen möchte muß ich sie normieren, das ist aber nur über die Leistung möglich, da ich keinen Parameter der Energie vom Hersteller habe. Insofern ist die Leistung spannender als die Energie. 
Anbei noch der Flow den ich zur Zeit nutze - vielleicht kann er ja bald weg 👍 
[flow.json](https://github.com/user-attachments/files/16284073/flow.json)

DerOetzi, 2024-07-18:
Naja aber wenn ein Modul z.b. in der Stunde 300 Wh Energie erzeugt hat ist die durchschnittliche Leistung entsprechend 300 W gewesen. Wenn du in 15 Minuten 75 Wh Erzeugung hast, dann sind das wohl 75 * 4 = 300 W. Bei einer Genauigkeit von 15 Minuten wirst du da kaum gravierende Abweichungen habe, weil  du spitzen usw schon rausgemittelt hast. Anders wäre es bei einer höheren zeitlichen Auflösung im Sekunden Bereich würde sich eine genauere Berechnung mittels integral erst lohnen

DerOetzi, 2024-07-18:
Achja und in dem flow ist leider keine API url enthalten, aber ich schau heute Abend Mal wenn die Kids im Bett sind!

skipperkm, 2024-07-19:
Hi,

Vielleicht habe ich mich falsch ausgedrückt, ich möchte da auch keine Integralrechnung auf ziehen. Es geht nur Um den Vergleich von Modulen unterschiedlicher Leistung. Die werden auf eine einheitliche Leistung (z.b. 400kWp) hoch oder runtergerechnet und sind dann direkt vergleichbar. D.h. ich sehe bei einem 375kwp Modul die Leistung die es hätte wenn es ein 400kWp Modul wäre. Dazu reichen auch die 15 min werte. Die sollten dann auch so bleiben wie sie kommen und nicht noch mit min,max und mean versehen werden - das brauchts da nicht mehr.

Mit freundlichen Grüßen 
😃

DerOetzi, 2024-07-19:
Okay ich hab gestern abend schon mal geschaut, sollte machbar sein über die Playback Daten (auch wenn ich mich bei dem fast JSON-Format, das da kommt, was den Entwickler geritten hat). Bin mir nur noch nicht ganz schlüssig wie ich das in die mqtt messages packe.

Die node-red erweiterung packt einfach immer den gesamten Tag in Viertelstundenintervallen raus, auch wenn dann für den Rest des Tages immer erstmal null geschrieben wird.

Gibt mehrere Optionen ich hab ja schon die einzelnen module wo ich den "aktuellen" Energiewert der einzelnen Module zurückgebe, da könnte ich das auch einfach so stumpf reinmappen. oder ich Teile das in zwei Topic-Gruppen

Und weitere Möglichkeit ich könnte es auch immer nur den Wert der letzten viertelstunde posten

DerOetzi, 2024-07-19:
wenn du das Docker Image mit dem tag `main` statt `latest` nimmst kannst du die Änderungen testen. Bitte um Rückmeldung, ob es so ist wie du dir es vorgestellt hast, dann kann ich auch einen Release drauss machen.

wegen den Spannungen usw. das wieder eine ganz andere Ecke. Die Auszuwerten, habe ich mal versucht, aber leider mit wenig Erfolg. Kann ich mir nochmal anschauen, wenn ich mit meinem anderem Projekt weitergekommen bin.

DerOetzi, 2024-07-19:
Achso die Änderung bewirkt, dass die PowerWerte pro Modul einmal in die InfluxDB geschoben werden alle 15 Minuten und auch mit im MQTT gepublished werden.

skipperkm, 2024-07-20:
Hi,

Du bist ja fix, danke. Ich schaue es mir an und melde mich. Muß jetzt erst mal in die Werkstatt.

Mit freundlichen Grüßen

skipperkm, 2024-07-22:
Guten Morgen,

hab mir das angeschaut. Auf Grund der der sich von Tag zu Tag ändernden Timestamps ist das ziemlich kompliziert auseiander zu frickeln. War wohl durch keine so gute Idee das stumpf reinzumappen.  Ich denke nur den wert der letzten 15 min zu übergeben reicht völlig aus. Das läßt sich besser weiter verarbeiten. In der Influx hat man es ja dann sowieso.

Mit freundlichen Grüßen
Schöne Woche

DerOetzi, 2024-07-22:
Hmm ich schau's mir Mal an, weil es könnte sein dass dann vielleicht ein Synchronisationsproblem entsteht, weil der Service immer nach seiner Systemzeit ziemlich genau um 0, 15, 30 und 45 den service-call macht und es dann manchmal sein kann das der SolarEdge Server erst um 16, 17 teilweise sogar erst 18 die Daten für die letzte Viertelstunde berechnet hat. 

Vielleicht wäre es eine lösung noch ein power_today zu ergänzen bei dem zusätzlich der komplette Tag aber nur die Uhrzeiten als Schlüssel ausgegeben wird. Was hälst du von dieser idee?

skipperkm, 2024-07-23:
Guten Morgen,

Das müsste ich mir mal anschauen wie das aussieht. Hätte jetzt auch nicht gedacht das das so aufwändig wird.
In dem Flow scheint das auch nicht viel anders umgesetzt zu sein. In so fern stellt sich die Frage ob man es so lässt.

Mit freundlichen Grüßen

DerOetzi, 2024-07-23:
Guten Abend,

habe mal das Feld power_today ergänzt. Würde es dann erstmal so lassen, es sei denn du hast noch eine bessere Idee.

Viele Grüße

skipperkm, 2024-07-24:
Guten Morgen,

ich schaue es mir nachher an. Und melde mich.

Mit freundlichen Grüßen

skipperkm, 2024-07-30:
Hi,
sorry, hab im moment viel um die Ohren - aber ja, ich habe es mir angesehen. Das sieht besser aus und ließe sich auch besser automatisieren als mit dem vollen Zeitstempel. Ich werde mal sehen ob ich das in eine Flow reinbekomme.
Und danke noch mal für die Mühe.

Mit freundlichen Grüßen

DerOetzi, 2024-08-27:
Ich würde dieses issue dann schließen @skipperkm

skipperkm, 2024-08-28:
Na klar, ist ok und danke noch mal Von meinem iPhone gesendet

closed 2024-08-28 by DerOetzi