DerOetzi/solaredge2mqtt #104: regular propanel error messges every night spam syslog

probutus, 2024-08-15:
Every night I get lots of propanel error messages with latest version:
Aug 15 01:11:16 andromeda propanel[6225]: IndexError: list index out of range
Aug 15 01:11:20 andromeda propanel[6225]: Task exception was never retrieved
Aug 15 01:11:20 andromeda propanel[6225]: future: <Task finished name='Task-5577004' coro=<EventBus._notify_listeners() done, defined at /usr/local/l>
Aug 15 01:11:20 andromeda propanel[6225]: Traceback (most recent call last):
Aug 15 01:11:20 andromeda propanel[6225]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/events/__init__.py", line 71, in _notif>
Aug 15 01:11:20 andromeda propanel[6225]:     await asyncio.gather(
Aug 15 01:11:20 andromeda propanel[6225]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/events/__init__.py", line 81, in _notif>
Aug 15 01:11:20 andromeda propanel[6225]:     await listener(event)
Aug 15 01:11:20 andromeda propanel[6225]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/powerflow/__init__.py", line 47, in>
Aug 15 01:11:20 andromeda propanel[6225]:     inverter_data, meters_data, batteries_data = await self.modbus.get_data()
Aug 15 01:11:20 andromeda propanel[6225]:                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Aug 15 01:11:20 andromeda propanel[6225]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/modbus/__init__.py", line 64, in ge>
Aug 15 01:11:20 andromeda propanel[6225]:     batteries_data = self._map_batteries(batteries_raw)
Aug 15 01:11:20 andromeda propanel[6225]:                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Aug 15 01:11:20 andromeda propanel[6225]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/modbus/__init__.py", line 150, in _>
Aug 15 01:11:20 andromeda propanel[6225]:     battery_data = SunSpecBattery(battery_raw)
Aug 15 01:11:20 andromeda propanel[6225]:                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
Aug 15 01:11:20 andromeda propanel[6225]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/modbus/models.py", line 145, in __i>
Aug 15 01:11:20 andromeda propanel[6225]:     status = BATTERY_STATUS_MAP[data["status"]]
Aug 15 01:11:20 andromeda propanel[6225]:              ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^

Please let me know if I should do some debugging or provide more information

DerOetzi, 2024-08-18:
Sorry for the late answer, I was a little bit busy last days! 

A first quick look says, it seems your battery reports an state the service doesn't know yet.

For further debbuging what kind of state this is I can do a patch with more debugging output. Is it possible with your setup to run the main branch instead of the release?

DerOetzi, 2024-08-18:
I have applied a patch to main branch which adds some exception handling. Please try it and send me the logs. There should be a Warning now instead of the stack trace, saying either

"Missing battery state"

or

"Invalid battery state: <INCORRECT STATE>"

probutus, 2024-08-18:
Hi,
thanks for the quick reply. I will install the latest mainline and post the log

probutus, 2024-08-18:
I noticed that the messages start at the point where the battery is empty and they stop again as soon as the battery gets charged from the inverter

DerOetzi, 2024-08-18:
I did some research on other libraries and found there is at least one missing state in battery states. I just pushed another version to main with this state "Sleeping". Please try this one.

probutus, 2024-08-19:
Hi, will do. The log with the debug output looks like this:

Aug 18 20:31:11 andromeda systemd[1]: Started solaredge2mqtt.service.
Aug 18 22:28:07 andromeda propanel[27822]: 2024-08-18 22:28:07.744 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop
Aug 18 22:28:33 andromeda propanel[27822]: 2024-08-18 22:28:33.890 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop
Aug 18 22:28:46 andromeda propanel[27822]: 2024-08-18 22:28:46.416 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop
Aug 18 22:28:50 andromeda propanel[27822]: 2024-08-18 22:28:50.749 | WARNING  | solaredge2mqtt.services.modbus.models:is_valid:174 - Battery state of charge is negative
Aug 18 22:28:50 andromeda propanel[27822]: 2024-08-18 22:28:50.750 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid battery data, skipping this loop
Aug 19 00:00:01 andromeda propanel[27822]: 2024-08-19 00:00:01.674 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - No energy data for today, skipping this loop
Aug 19 00:13:20 andromeda propanel[27822]: 2024-08-19 00:13:20.963 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Value change not valid, skipping this loop, skipping this >
Aug 19 01:25:36 andromeda propanel[27822]: 2024-08-19 01:25:36.076 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid battery status: 10, skipping this loop
Aug 19 01:25:40 andromeda propanel[27822]: 2024-08-19 01:25:40.810 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid battery status: 10, skipping this loop
Aug 19 01:25:46 andromeda propanel[27822]: 2024-08-19 01:25:46.025 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid battery status: 10, skipping this loop
Aug 19 01:25:51 andromeda propanel[27822]: 2024-08-19 01:25:51.083 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid battery status: 10, skipping this loop
Aug 19 01:25:55 andromeda propanel[27822]: 2024-08-19 01:25:55.986 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid battery status: 10, skipping this loop
Aug 19 01:26:00 andromeda propanel[27822]: 2024-08-19 01:26:00.912 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid battery status: 10, skipping this loop
A
.....
I will now change to the latest version

probutus, 2024-08-20:
With the latest version the error messages have drastically decreased:
Aug 19 06:25:28 andromeda systemd[1]: Started solaredge2mqtt.service.
Aug 19 07:40:45 andromeda propanel[13580]: 2024-08-19 07:40:45.874 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Value change not valid, skipping this loop, skipping this loop
Aug 19 08:59:15 andromeda propanel[13580]: 2024-08-19 08:59:15.987 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Value change not valid, skipping this loop, skipping this loop
Aug 19 09:31:31 andromeda propanel[13580]: 2024-08-19 09:31:31.176 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Value change not valid, skipping this loop, skipping this loop
Aug 19 09:49:26 andromeda propanel[13580]: 2024-08-19 09:49:26.443 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Value change not valid, skipping this loop, skipping this loop
Aug 19 21:58:30 andromeda propanel[13580]: 2024-08-19 21:58:30.596 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Value change not valid, skipping this loop, skipping this loop
Aug 19 22:32:52 andromeda propanel[13580]: 2024-08-19 22:32:52.701 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop
Aug 19 22:33:18 andromeda propanel[13580]: 2024-08-19 22:33:18.846 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop
Aug 19 22:33:33 andromeda propanel[13580]: 2024-08-19 22:33:33.015 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop
Aug 19 22:33:35 andromeda propanel[13580]: 2024-08-19 22:33:35.338 | WARNING  | solaredge2mqtt.services.modbus.models:is_valid:187 - Battery state of charge is negative
Aug 19 22:33:35 andromeda propanel[13580]: 2024-08-19 22:33:35.338 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid battery data, skipping this loop
Aug 19 22:33:40 andromeda propanel[13580]: 2024-08-19 22:33:40.333 | WARNING  | solaredge2mqtt.services.modbus.models:is_valid:187 - Battery state of charge is negative
Aug 19 22:33:40 andromeda propanel[13580]: 2024-08-19 22:33:40.333 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid battery data, skipping this loop
Aug 19 23:47:15 andromeda propanel[13580]: 2024-08-19 23:47:15.669 | WARNING  | solaredge2mqtt.services.powerflow.models:is_valid:76 - Consumer used production + grid delivery is not equal to inverter production
Aug 19 23:47:15 andromeda propanel[13580]: 2024-08-19 23:47:15.670 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid powerflow data, skipping this loop

DerOetzi, 2024-08-27:
Okay I'm working on another version at the moment. There I replace the outdated solaredge_modbus library which seems not to be maintained anymore. Have to do some internal tests and will push to the end of the week to the main branch. 

Some information to the warnings you see:

 - Value change not valid, skipping this loop, skipping this loop
 - Consumer used production + grid delivery is not equal to inverter production
 
 This is some kind of debouncing and filtering I introduced to the service, because the SolarEdge Components reporting sometimes "invalid". For example my inverter sometimes reports at night suddently a production of 10 kW for one request loop. The first of those error messages looks like your inverter does the same.  And sometimes a measurement loop hits a ms the inverter data is already changed but the meter isn't. Then you have such strange powerflow values of second warning

- Invalid modbus data, skipping this loop

Can you investigate a little more here is there any other stacktrace or message before, maybe your inverter wasn't reachable for some reason

-  Battery state of charge is negative

This is strange why your battery reports negative state of charge.

probutus, 2024-08-29:
Many thanks. I will download a new version over the weekend and give it a try. We can close this topic

closed 2024-08-29 by probutus