DerOetzi/solaredge2mqtt #117: lots of exceptions happening in solaredge2mqtt

probutus, 2024-11-23:
Hello,

I have the following error messages in my log with the latest version:
Nov 23 16:41:36 andromeda propanel[2509]: Task exception was never retrieved
Nov 23 16:41:36 andromeda propanel[2509]: future: <Task finished name='Task-1424248' coro=<EventBus._notify_listeners() done, defined at /usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/events/__init__.py:75> exception=1 validation error for GridPowerflow
Nov 23 16:41:36 andromeda propanel[2509]: power
Nov 23 16:41:36 andromeda propanel[2509]:   Input should be a valid integer, got a number with a fractional part [type=int_from_float, input_value=-6.42, input_type=float]
Nov 23 16:41:36 andromeda propanel[2509]:     For further information visit https://errors.pydantic.dev/2.8/v/int_from_float>
Nov 23 16:41:36 andromeda propanel[2509]: Traceback (most recent call last):
Nov 23 16:41:36 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/events/__init__.py", line 79, in _notify_listeners
Nov 23 16:41:36 andromeda propanel[2509]:     await asyncio.gather(
Nov 23 16:41:36 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/events/__init__.py", line 89, in _notify_listener
Nov 23 16:41:36 andromeda propanel[2509]:     await listener(event)
Nov 23 16:41:36 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/powerflow/__init__.py", line 69, in calculate_powerflow
Nov 23 16:41:36 andromeda propanel[2509]:     powerflow = Powerflow.from_modbus(
Nov 23 16:41:36 andromeda propanel[2509]:                 ^^^^^^^^^^^^^^^^^^^^^^
Nov 23 16:41:36 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/powerflow/models.py", line 44, in from_modbus
Nov 23 16:41:36 andromeda propanel[2509]:     grid = GridPowerflow.from_modbus(meters_data)
Nov 23 16:41:36 andromeda propanel[2509]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 23 16:41:36 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/powerflow/models.py", line 217, in from_modbus
Nov 23 16:41:36 andromeda propanel[2509]:     return GridPowerflow(power=grid)
Nov 23 16:41:36 andromeda propanel[2509]:            ^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 23 16:41:36 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/pydantic/main.py", line 193, in __init__
Nov 23 16:41:36 andromeda propanel[2509]:     self.__pydantic_validator__.validate_python(data, self_instance=self)
Nov 23 16:41:36 andromeda propanel[2509]: pydantic_core._pydantic_core.ValidationError: 1 validation error for GridPowerflow
Nov 23 16:41:36 andromeda propanel[2509]: power
Nov 23 16:41:36 andromeda propanel[2509]:   Input should be a valid integer, got a number with a fractional part [type=int_from_float, input_value=-6.42, input_type=float]
Nov 23 16:41:36 andromeda propanel[2509]:     For further information visit https://errors.pydantic.dev/2.8/v/int_from_float
Nov 23 16:41:41 andromeda propanel[2509]: Task exception was never retrieved
Nov 23 16:41:41 andromeda propanel[2509]: future: <Task finished name='Task-1424250' coro=<EventBus._notify_listeners() done, defined at /usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/events/__init__.py:75> exception=1 validation error for GridPowerflow
Nov 23 16:41:41 andromeda propanel[2509]: power
Nov 23 16:41:41 andromeda propanel[2509]:   Input should be a valid integer, got a number with a fractional part [type=int_from_float, input_value=11.49, input_type=float]
Nov 23 16:41:41 andromeda propanel[2509]:     For further information visit https://errors.pydantic.dev/2.8/v/int_from_float>
Nov 23 16:41:41 andromeda propanel[2509]: Traceback (most recent call last):
Nov 23 16:41:41 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/events/__init__.py", line 79, in _notify_listeners
Nov 23 16:41:41 andromeda propanel[2509]:     await asyncio.gather(
Nov 23 16:41:41 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/events/__init__.py", line 89, in _notify_listener
Nov 23 16:41:41 andromeda propanel[2509]:     await listener(event)
Nov 23 16:41:41 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/powerflow/__init__.py", line 69, in calculate_powerflow
Nov 23 16:41:41 andromeda propanel[2509]:     powerflow = Powerflow.from_modbus(
Nov 23 16:41:41 andromeda propanel[2509]:                 ^^^^^^^^^^^^^^^^^^^^^^
Nov 23 16:41:41 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/powerflow/models.py", line 44, in from_modbus
Nov 23 16:41:41 andromeda propanel[2509]:     grid = GridPowerflow.from_modbus(meters_data)
Nov 23 16:41:41 andromeda propanel[2509]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 23 16:41:41 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/powerflow/models.py", line 217, in from_modbus
Nov 23 16:41:41 andromeda propanel[2509]:     return GridPowerflow(power=grid)
Nov 23 16:41:41 andromeda propanel[2509]:            ^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 23 16:41:41 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/pydantic/main.py", line 193, in __init__
Nov 23 16:41:41 andromeda propanel[2509]:     self.__pydantic_validator__.validate_python(data, self_instance=self)
Nov 23 16:41:41 andromeda propanel[2509]: pydantic_core._pydantic_core.ValidationError: 1 validation error for GridPowerflow
Nov 23 16:41:41 andromeda propanel[2509]: power
Nov 23 16:41:41 andromeda propanel[2509]:   Input should be a valid integer, got a number with a fractional part [type=int_from_float, input_value=11.49, input_type=float]
Nov 23 16:41:41 andromeda propanel[2509]:     For further information visit https://errors.pydantic.dev/2.8/v/int_from_float
Nov 23 16:41:46 andromeda propanel[2509]: Task exception was never retrieved
Nov 23 16:41:46 andromeda propanel[2509]: future: <Task finished name='Task-1424252' coro=<EventBus._notify_listeners() done, defined at /usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/events/__init__.py:75> exception=1 validation error for GridPowerflow
Nov 23 16:41:46 andromeda propanel[2509]: power
Nov 23 16:41:46 andromeda propanel[2509]:   Input should be a valid integer, got a number with a fractional part [type=int_from_float, input_value=-16.99, input_type=float]
Nov 23 16:41:46 andromeda propanel[2509]:     For further information visit https://errors.pydantic.dev/2.8/v/int_from_float>
Nov 23 16:41:46 andromeda propanel[2509]: Traceback (most recent call last):
Nov 23 16:41:46 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/events/__init__.py", line 79, in _notify_listeners
Nov 23 16:41:46 andromeda propanel[2509]:     await asyncio.gather(
Nov 23 16:41:46 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/events/__init__.py", line 89, in _notify_listener
Nov 23 16:41:46 andromeda propanel[2509]:     await listener(event)
Nov 23 16:41:46 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/powerflow/__init__.py", line 69, in calculate_powerflow
Nov 23 16:41:46 andromeda propanel[2509]:     powerflow = Powerflow.from_modbus(
Nov 23 16:41:46 andromeda propanel[2509]:                 ^^^^^^^^^^^^^^^^^^^^^^
Nov 23 16:41:46 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/powerflow/models.py", line 44, in from_modbus
Nov 23 16:41:46 andromeda propanel[2509]:     grid = GridPowerflow.from_modbus(meters_data)
Nov 23 16:41:46 andromeda propanel[2509]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 23 16:41:46 andromeda propanel[2509]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/powerflow/models.py", line 217, in from_modbus
[... 110 lines cut ...]
Nov 23 16:42:11 andromeda propanel[2509]: power
Nov 23 16:42:11 andromeda propanel[2509]:   Input should be a valid integer, got a number with a fractional part [type=int_from_float, input_value=-0.61, input_type=float]
Nov 23 16:42:11 andromeda propanel[2509]:     For further information visit https://errors.pydantic.dev/2.8/v/int_from_float>
Nov 23 16:42:11 andromeda propanel[2509]: Traceback (most recent call last):
[... cut ...]
Nov 23 16:42:11 andromeda propanel[2509]: pydantic_core._pydantic_core.ValidationError: 1 validation error for GridPowerflow
Nov 23 16:42:11 andromeda propanel[2509]: power
Nov 23 16:42:11 andromeda propanel[2509]:   Input should be a valid integer, got a number with a fractional part [type=int_from_float, input_value=-0.61, input_type=float]
Nov 23 16:42:11 andromeda propanel[2509]:     For further information visit https://errors.pydantic.dev/2.8/v/int_from_float

Please let me know if I can provide further debug information

DerOetzi, 2024-11-25:
Which kind of meter do you have installed to track the export+import? And which version sof se2mqtt are you running?

probutus, 2024-11-25:
I have an SE9k 3 phase inverter together with an SE MTR240-3PC. This is inside the SE9K inverter. I use the latest se2mqtt version

DerOetzi, 2025-01-14:
@probutus I think I have good news. I had the same error messages at my site around christmas and think I have found a fix. Can you use main-branch version instead of latest release version and verify it is fixed on your site as well, before I do a release with it.

probutus, 2025-01-15:
Hi, 
happy new year and many thanks for the update. I tried the patch and it solves 98% of the problematic output. This is definitely a big improvement!

There is only one strange thing left. Every 10 minutes (at exactly the time offset, e.g. 10:20, 10:30 ...) I get the exception below:

Best regards and again many thanks!
Alex

Jan 15 04:20:11 andromeda propanel[18348]: Task exception was never retrieved
Jan 15 04:20:11 andromeda propanel[18348]: future: <Task finished name='Task-6949' coro=<EventBus._notify_listeners() done, defined at /usr/local/lib/python3.11/dist-packages/solaredge2mq>
Jan 15 04:20:11 andromeda propanel[18348]: Traceback (most recent call last):
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 450, in _make_request
Jan 15 04:20:11 andromeda propanel[18348]:     six.raise_from(e, None)
Jan 15 04:20:11 andromeda propanel[18348]:   File "<string>", line 3, in raise_from
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 445, in _make_request
Jan 15 04:20:11 andromeda propanel[18348]:     httplib_response = conn.getresponse()
Jan 15 04:20:11 andromeda propanel[18348]:                        ^^^^^^^^^^^^^^^^^^
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/lib/python3.11/http/client.py", line 1374, in getresponse
Jan 15 04:20:11 andromeda propanel[18348]:     response.begin()
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/lib/python3.11/http/client.py", line 318, in begin
Jan 15 04:20:11 andromeda propanel[18348]:     version, status, reason = self._read_status()
Jan 15 04:20:11 andromeda propanel[18348]:                               ^^^^^^^^^^^^^^^^^^^
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/lib/python3.11/http/client.py", line 279, in _read_status
Jan 15 04:20:11 andromeda propanel[18348]:     line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
Jan 15 04:20:11 andromeda propanel[18348]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/lib/python3.11/socket.py", line 706, in readinto
Jan 15 04:20:11 andromeda propanel[18348]:     return self._sock.recv_into(b)
Jan 15 04:20:11 andromeda propanel[18348]:            ^^^^^^^^^^^^^^^^^^^^^^^
Jan 15 04:20:11 andromeda propanel[18348]: TimeoutError: timed out
Jan 15 04:20:11 andromeda propanel[18348]: During handling of the above exception, another exception occurred:
Jan 15 04:20:11 andromeda propanel[18348]: Traceback (most recent call last):
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/events/__init__.py", line 79, in _notify_listeners
Jan 15 04:20:11 andromeda propanel[18348]:     await asyncio.gather(
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/events/__init__.py", line 89, in _notify_listener
Jan 15 04:20:11 andromeda propanel[18348]:     await listener(event)
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/influxdb/__init__.py", line 101, in loop
Jan 15 04:20:11 andromeda propanel[18348]:     self.delete_from_measurements(
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/influxdb/__init__.py", line 114, in delete_from_measurements
Jan 15 04:20:11 andromeda propanel[18348]:     self.delete_from_measurement(start, stop, measurement)
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/influxdb/__init__.py", line 119, in delete_from_measurement
Jan 15 04:20:11 andromeda propanel[18348]:     self.delete_api.delete(
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/local/lib/python3.11/dist-packages/influxdb_client/client/delete_api.py", line 35, in delete
Jan 15 04:20:11 andromeda propanel[18348]:     return self._service.post_delete(delete_predicate_request=predicate_request, bucket=bucket, org=org_param)
Jan 15 04:20:11 andromeda propanel[18348]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/local/lib/python3.11/dist-packages/influxdb_client/service/delete_service.py", line 56, in post_delete
Jan 15 04:20:11 andromeda propanel[18348]:     (data) = self.post_delete_with_http_info(delete_predicate_request, **kwargs)  # noqa: E501
Jan 15 04:20:11 andromeda propanel[18348]:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/local/lib/python3.11/dist-packages/influxdb_client/service/delete_service.py", line 82, in post_delete_with_http_info
Jan 15 04:20:11 andromeda propanel[18348]:     return self.api_client.call_api(
Jan 15 04:20:11 andromeda propanel[18348]:            ^^^^^^^^^^^^^^^^^^^^^^^^^
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/local/lib/python3.11/dist-packages/influxdb_client/_sync/api_client.py", line 343, in call_api
Jan 15 04:20:11 andromeda propanel[18348]:     return self.__call_api(resource_path, method,
Jan 15 04:20:11 andromeda propanel[18348]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/local/lib/python3.11/dist-packages/influxdb_client/_sync/api_client.py", line 173, in __call_api
[... cut ...]
Jan 15 04:20:11 andromeda propanel[18348]:     self._raise_timeout(err=e, url=url, timeout_value=read_timeout)
Jan 15 04:20:11 andromeda propanel[18348]:   File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 341, in _raise_timeout
Jan 15 04:20:11 andromeda propanel[18348]:     raise ReadTimeoutError(
Jan 15 04:20:11 andromeda propanel[18348]: urllib3.exceptions.ReadTimeoutError: HTTPConnectionPool(host='192.168.1.5', port=8086): Read timed out. (read timeout=9.999651651829481)

DerOetzi, 2025-01-16:
This can be mainly two things therefore I need to know a bit more about your configuration.

As I see you use the influxdb functionality to store timeseries data:
- Can you please check the access token you use has full rights.
- Do you have prices for incoming and outgoing configured?
 Because it aggregates the powerflow_raw data to energy_data and money_data for current and last hour every 10 minutes. And afterwards he applies a kind of retention by deleting raw data older then 24 hours.

Do you try to use the forecast functionality, because this would be triggered although afterwards?

probutus, 2025-01-16:
Thanks for the quick reply.
* Access token has full rights (read+write)
* Forecast is not enabled
* Prices and consumption are enabled
* webscraping from SE website (monitoring site) with user+pass is enabled

But I get all 5 seconds (poll interval) that SE2MQTT received invalid data and is skipping the loop

Startup log is this:
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.251 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: interval10mintriggerevent
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.278 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: influxdbaggregatedevent
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.279 | INFO     | solaredge2mqtt.services.modbus:__init__:46 - Using SolarEdge inverter via modbus: 192.168.1.88:1502
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.280 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: intervalbasetriggerevent
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.280 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: interval15mintriggerevent
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.281 | INFO     | solaredge2mqtt.services.homeassistant:__init__:37 - Home Assistant discovery enabled
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.281 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: forecastevent
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.282 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: energyreadevent
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.282 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: modbusinverterreadevent
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.283 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: powerflowgeneratedevent
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.283 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: wallboxreadevent
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.283 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: modbusmetersreadevent
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.284 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: modbusbatteriesreadevent
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.284 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: mqttreceivedevent
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.286 | INFO     | solaredge2mqtt.service:main_loop:114 - Starting SolarEdge2MQTT service...
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.287 | INFO     | solaredge2mqtt.service:main_loop:115 - Version: 1.1.3+19.g612d358
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.290 | INFO     | solaredge2mqtt.service:main_loop:116 - Operationg system: Linux-6.1.70-v8+-aarch64-with-glibc2.36 (Linux/aarch64)
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.291 | INFO     | solaredge2mqtt.service:main_loop:120 - Timezone: Europe/Berlin
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.297 | INFO     | solaredge2mqtt.core.influxdb:initialize_buckets:80 - Bucket 'solaredge' already exists.
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.298 | INFO     | solaredge2mqtt.core.mqtt:__init__:21 - Using MQTT broker: 192.168.1.5:1883
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.298 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: mqttpublishevent
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.299 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: mqttsubscribeevent
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.304 | INFO     | solaredge2mqtt.core.mqtt:_subscribe_topic:52 - Subscribing to topic: homeassistant/status
Jan 16 19:56:39 andromeda propanel[6088]: 2025-01-16 19:56:39.306 | INFO     | solaredge2mqtt.core.mqtt:listen:58 - MQTT listen on subscribed topics

Jan 16 19:56:41 andromeda propanel[6088]: 2025-01-16 19:56:41.311 | INFO     | solaredge2mqtt.services.modbus:_block_register_slice:165 - Block unreadable registers beginning at 40000
Jan 16 19:56:42 andromeda propanel[6088]: 2025-01-16 19:56:42.314 | INFO     | solaredge2mqtt.services.modbus:_block_register_slice:165 - Block unreadable registers beginning at 40188
Jan 16 19:56:43 andromeda propanel[6088]: 2025-01-16 19:56:43.317 | INFO     | solaredge2mqtt.services.modbus:_block_register_slice:165 - Block unreadable registers beginning at 40362
Jan 16 19:56:44 andromeda propanel[6088]: 2025-01-16 19:56:44.320 | INFO     | solaredge2mqtt.services.modbus:_block_register_slice:165 - Block unreadable registers beginning at 40536
Jan 16 19:56:45 andromeda propanel[6088]: 2025-01-16 19:56:45.322 | INFO     | solaredge2mqtt.services.modbus:_block_register_slice:165 - Block unreadable registers beginning at 57664
Jan 16 19:56:46 andromeda propanel[6088]: 2025-01-16 19:56:46.325 | INFO     | solaredge2mqtt.services.modbus:_block_register_slice:165 - Block unreadable registers beginning at 57920
Jan 16 19:56:47 andromeda propanel[6088]: 2025-01-16 19:56:47.328 | INFO     | solaredge2mqtt.services.modbus:_block_register_slice:165 - Block unreadable registers beginning at 61440
[... cut ...]
Jan 16 19:57:35 andromeda propanel[6088]: 2025-01-16 19:57:35.416 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop
Jan 16 19:57:40 andromeda propanel[6088]: 2025-01-16 19:57:40.423 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop
Jan 16 19:57:45 andromeda propanel[6088]: 2025-01-16 19:57:45.430 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop

probutus, 2025-01-20:
I have an update to the exceptions.
I created a new bucket for solaredge2mqtt to collect the data and the execptions are gone.
When setting up solaredge2mqtt. I first had forecast functionality disabled and enabled it later on along with the website data scraping. Maybe there is a correlation for the later enabling of new data collection with the exception

DerOetzi, 2025-01-30:
Okay, that is indeed a strange behavior. I added both features later and enabled them afterward, not from the beginning. Especially the data scraping feature is quite "new."

Yes, I rewrote the entire Modbus reading from scratch because the solaredgemodbus library I was using relied on deprecated components. It's possible that I missed something during the migration. I apologize for any issues this may have caused.

Currently, I'm still in the process of transferring my smart home setup from OpenHAB to HASS, so I won't have much time to investigate this in the coming weeks. However, feel free to send me some example logs from se2mqtt.

probutus, 2025-02-01:
Hi,
thanks for taking care! I opened a dedicated issue for the modbusproxy issue including the logs. So I close this issue for now as there are no more exceptions anymore

closed 2025-02-01 by probutus