DerOetzi/solaredge2mqtt #127: Sync methods (influxdb, monitoring site request)  has conflicts with new async modbus calls

DerOetzi, 2025-02-14:
The new introduced async modbus has regular issues and reporting
Invalid modbus data, skipping loops

The problems are the blocking service calls for example influxdb and monitoring site which let async modbus calls running into timeout every 10 Minutes and 15 Minutes

Solution all services have to be refactored to async

DerOetzi, 2025-02-24:
@probotus May I ask you to pull the latest version of the image tag ghcr.io/deroetzi/solaredge2mqtt:replace_solaredge_modbus and verify that there are hopfully no problems with the async modbus version anymore?

probutus, 2025-02-25:
I tested the new version for 24h. It started with sporadic can not read register requests

```
Feb 24 06:01:37 andromeda propanel[9625]: No response received after 1 retries, continue with next request
Feb 24 06:01:37 andromeda propanel[9625]: 2025-02-24 06:01:37.106 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 40071
Feb 24 06:01:39 andromeda propanel[9625]: No response received after 1 retries, continue with next request
Feb 24 06:01:39 andromeda propanel[9625]: 2025-02-24 06:01:39.110 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 40190
Feb 24 06:01:41 andromeda propanel[9625]: No response received after 1 retries, continue with next request
Feb 24 06:01:41 andromeda propanel[9625]: 2025-02-24 06:01:41.114 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 57712
Feb 24 06:01:41 andromeda propanel[9625]: 2025-02-24 06:01:41.115 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop
Feb 24 06:01:41 andromeda propanel[9625]: Cancel send, because not connected!
Feb 24 06:01:42 andromeda propanel[9625]: Cancel send, because not connected!
```
after several hours I got loads of exceptions until 19:47 when the regular can not read register comes back:

[output.txt.gz](https://github.com/user-attachments/files/18971352/output.txt.gz)

`Feb 25 12:11:53 andromeda propanel[10047]: 2025-02-25 12:11:53.473 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop
Feb 25 12:11:55 andromeda propanel[10047]: 2025-02-25 12:11:55.396 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 40190
Feb 25 12:11:55 andromeda propanel[10047]: Task exception was never retrieved
Feb 25 12:11:55 andromeda propanel[10047]: future: <Task finished name='Task-120280' coro=<EventBus._notify_listeners() done, defined at /usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/events/__init__.py:75> exception=5 validation errors for SunSpecACVoltage
Feb 25 12:11:55 andromeda propanel[10047]: l1
Feb 25 12:11:55 andromeda propanel[10047]:   Input should be a valid number [type=float_type, input_value=2291000000000000000000000...000000000000000000000000, input_type=int]
Feb 25 12:11:55 andromeda propanel[10047]:     For further information visit https://errors.pydantic.dev/2.8/v/float_type
Feb 25 12:11:55 andromeda propanel[10047]: l2
Feb 25 12:11:55 andromeda propanel[10047]:   Input should be a valid number [type=float_type, input_value=2291000000000000000000000...000000000000000000000000, input_type=int]
Feb 25 12:11:55 andromeda propanel[10047]:     For further information visit https://errors.pydantic.dev/2.8/v/float_type
Feb 25 12:11:55 andromeda propanel[10047]: l3
Feb 25 12:11:55 andromeda propanel[10047]:   Input should be a valid number [type=float_type, input_value=2284000000000000000000000...000000000000000000000000, input_type=int]
Feb 25 12:11:55 andromeda propanel[10047]:     For further information visit https://errors.pydantic.dev/2.8/v/float_type
Feb 25 12:11:55 andromeda propanel[10047]: l1n
Feb 25 12:11:55 andromeda propanel[10047]:   Input should be a valid number [type=float_type, input_value=2288000000000000000000000...000000000000000000000000, input_type=int]
Feb 25 12:11:55 andromeda propanel[10047]:     For further information visit https://errors.pydantic.dev/2.8/v/float_type
Feb 25 12:11:55 andromeda propanel[10047]: l3n
Feb 25 12:11:55 andromeda propanel[10047]:   Input should be a valid number [type=float_type, input_value=3962000000000000000000000...000000000000000000000000, input_type=int]
Feb 25 12:11:55 andromeda propanel[10047]:     For further information visit https://errors.pydantic.dev/2.8/v/float_type>
Feb 25 12:11:55 andromeda propanel[10047]: Traceback (most recent call last):
Feb 25 12:11:55 andromeda propanel[10047]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/events/__init__.py", line 79, in _notify_listeners
Feb 25 12:11:55 andromeda propanel[10047]:     await asyncio.gather(
Feb 25 12:11:55 andromeda propanel[10047]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/core/events/__init__.py", line 89, in _notify_listener
Feb 25 12:11:55 andromeda propanel[10047]:     await listener(event)
Feb 25 12:11:55 andromeda propanel[10047]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/powerflow/__init__.py", line 51, in calculate_powerflow
Feb 25 12:11:55 andromeda propanel[10047]:     inverter_data, meters_data, batteries_data = await self.modbus.get_data()
Feb 25 12:11:55 andromeda propanel[10047]:                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 25 12:11:55 andromeda propanel[10047]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/modbus/__init__.py", line 144, in get_data
Feb 25 12:11:55 andromeda propanel[10047]:     inverter_data = self._map_inverter(inverter_raw)
Feb 25 12:11:55 andromeda propanel[10047]:                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 25 12:11:55 andromeda propanel[10047]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/modbus/__init__.py", line 233, in _map_inverter
Feb 25 12:11:55 andromeda propanel[10047]:     inverter_data = SunSpecInverter(self._device_info["inverter"], inverter_raw)
[... cut ...]
Feb 25 12:11:55 andromeda propanel[10047]:     inverter_data = SunSpecInverter(self._device_info["inverter"], inverter_raw)
Feb 25 12:11:55 andromeda propanel[10047]:                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 25 12:11:55 andromeda propanel[10047]:   File "/usr/local/lib/python3.11/dist-packages/solaredge2mqtt/services/modbus/models.py", line 96, in __init__
Feb 25 12:11:55 andromeda propanel[10047]:     ac = SunSpecAC(data)
Feb 25 12:11:55 andromeda propanel[10047]:          ^^^^^^^^^^^^^^^

probutus, 2025-02-26:
todays errorlog was much shorter. Looks like my inverter had a bad day yesterday

`
root@andromeda:/home/pi# journalctl -u solaredge2mqtt --since today
Feb 26 05:50:02 andromeda propanel[10047]: No response received after 1 retries, continue with next request
Feb 26 05:50:02 andromeda propanel[10047]: 2025-02-26 05:50:02.493 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 40071
Feb 26 05:50:04 andromeda propanel[10047]: No response received after 1 retries, continue with next request
Feb 26 05:50:04 andromeda propanel[10047]: 2025-02-26 05:50:04.497 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 40190
Feb 26 05:50:06 andromeda propanel[10047]: No response received after 1 retries, continue with next request
Feb 26 05:50:06 andromeda propanel[10047]: 2025-02-26 05:50:06.501 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 57712
Feb 26 05:50:06 andromeda propanel[10047]: 2025-02-26 05:50:06.502 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop
Feb 26 05:50:06 andromeda propanel[10047]: Cancel send, because not connected!
Feb 26 05:50:07 andromeda propanel[10047]: Cancel send, because not connected!
Feb 26 05:50:08 andromeda propanel[10047]: No response received after 1 retries, continue with next request
Feb 26 05:50:08 andromeda propanel[10047]: 2025-02-26 05:50:08.508 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 40071
Feb 26 05:50:08 andromeda propanel[10047]: 2025-02-26 05:50:08.509 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 40190
Feb 26 05:50:08 andromeda propanel[10047]: 2025-02-26 05:50:08.509 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 57712
Feb 26 05:50:08 andromeda propanel[10047]: 2025-02-26 05:50:08.510 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop
Feb 26 05:50:12 andromeda propanel[10047]: No response received after 1 retries, continue with next request
Feb 26 05:50:12 andromeda propanel[10047]: 2025-02-26 05:50:12.509 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 40071
Feb 26 05:50:14 andromeda propanel[10047]: No response received after 1 retries, continue with next request
Feb 26 05:50:14 andromeda propanel[10047]: 2025-02-26 05:50:14.512 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 40190
Feb 26 05:50:16 andromeda propanel[10047]: No response received after 1 retries, continue with next request
Feb 26 05:50:16 andromeda propanel[10047]: 2025-02-26 05:50:16.515 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 57712
Feb 26 05:50:16 andromeda propanel[10047]: 2025-02-26 05:50:16.516 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop
Feb 26 05:50:16 andromeda propanel[10047]: Cancel send, because not connected!
Feb 26 05:50:17 andromeda propanel[10047]: Cancel send, because not connected!
Feb 26 05:50:18 andromeda propanel[10047]: No response received after 1 retries, continue with next request
Feb 26 05:50:18 andromeda propanel[10047]: 2025-02-26 05:50:18.521 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 40071
Feb 26 05:50:18 andromeda propanel[10047]: 2025-02-26 05:50:18.522 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 40190
Feb 26 05:50:18 andromeda propanel[10047]: 2025-02-26 05:50:18.522 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 57712
Feb 26 05:50:18 andromeda propanel[10047]: 2025-02-26 05:50:18.523 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop
Feb 26 05:50:22 andromeda propanel[10047]: No response received after 1 retries, continue with next request
Feb 26 05:50:22 andromeda propanel[10047]: 2025-02-26 05:50:22.525 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 40071
Feb 26 05:50:24 andromeda propanel[10047]: No response received after 1 retries, continue with next request
Feb 26 05:50:24 andromeda propanel[10047]: 2025-02-26 05:50:24.529 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 40190
Feb 26 05:50:26 andromeda propanel[10047]: No response received after 1 retries, continue with next request
Feb 26 05:50:26 andromeda propanel[10047]: 2025-02-26 05:50:26.533 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 57712
Feb 26 05:50:26 andromeda propanel[10047]: 2025-02-26 05:50:26.533 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop
Feb 26 05:50:26 andromeda propanel[10047]: Cancel send, because not connected!
Feb 26 05:50:27 andromeda propanel[10047]: Cancel send, because not connected!
Feb 26 05:50:28 andromeda propanel[10047]: No response received after 1 retries, continue with next request
Feb 26 05:50:28 andromeda propanel[10047]: 2025-02-26 05:50:28.536 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:217 - Unreadable register 40071
[... cut ...]
Feb 26 19:54:06 andromeda propanel[10047]: 2025-02-26 19:54:06.053 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid battery data, skipping this loop
Feb 26 19:54:10 andromeda propanel[10047]: 2025-02-26 19:54:10.738 | WARNING  | solaredge2mqtt.services.modbus.models:is_valid:191 - Battery state of charge is negative
Feb 26 19:54:10 andromeda propanel[10047]: 2025-02-26 19:54:10.738 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid battery data, skipping this loop

`

DerOetzi, 2025-04-30:
Should be fixed with v1.3.0 changes

closed 2025-04-30 by DerOetzi