DerOetzi/solaredge2mqtt #152: slave inverter does not work correctly

mihadoo, 2025-05-25:
Hallo Oetzi

I have 2 SE inverters connected over modbus. The master inverter is available on modbus id 1 and solaredge2mqtt reads all values incl. battery correctly. The slave inverter is available on modbus id 2 on the IP of the master inverter. When I try to read values for ID2 (for slave inverter) it does not work with solaredge2mqtt. It works OK when I read values using openHAB but I would prefer having all data on MQTT as JSON:

```
2025-05-25 07:22:15.677 | INFO     | solaredge2mqtt.services.modbus:__init__:47 - Using SolarEdge inverter via modbus: 192.168.218.21:9011
2025-05-25 07:22:15.677 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: intervalbasetriggerevent
2025-05-25 07:22:15.679 | INFO     | solaredge2mqtt.service:main_loop:115 - Starting SolarEdge2MQTT service...
2025-05-25 07:22:15.679 | INFO     | solaredge2mqtt.service:main_loop:116 - Version: 1.4.1
2025-05-25 07:22:15.682 | INFO     | solaredge2mqtt.service:main_loop:117 - Operationg system: Linux-6.1.21-v8+-aarch64-with-glibc2.31 (Linux/aarch64)
2025-05-25 07:22:15.683 | INFO     | solaredge2mqtt.service:main_loop:121 - Timezone: Europe/Ljubljana
2025-05-25 07:22:15.683 | INFO     | solaredge2mqtt.core.mqtt:__init__:21 - Using MQTT broker: 192.168.218.21:1883
2025-05-25 07:22:15.683 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: mqttpublishevent
2025-05-25 07:22:15.683 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: mqttsubscribeevent
2025-05-25 07:22:15.688 | INFO     | solaredge2mqtt.services.modbus:async_init:63 - Initializing modbus
2025-05-25 07:22:21.226 | INFO     | solaredge2mqtt.services.modbus:read_device_info:123 - Found inverter SolarEdge SE12.5K-RW0T0BNN4 7E1D8185
Traceback (most recent call last):
  File "/home/openhabian/.local/bin/solaredge2mqtt", line 8, in <module>
    sys.exit(run())
             ~~~^^
  File "/home/openhabian/.local/lib/python3.13/site-packages/solaredge2mqtt/service.py", line 41, in run
    loop.run_until_complete(service.main_loop())
    ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 719, in run_until_complete
    return future.result()
           ~~~~~~~~~~~~~^^
  File "/home/openhabian/.local/lib/python3.13/site-packages/solaredge2mqtt/service.py", line 136, in main_loop
    await self.powerflow.async_init()
  File "/home/openhabian/.local/lib/python3.13/site-packages/solaredge2mqtt/services/powerflow/__init__.py", line 48, in async_init
    await self.modbus.async_init()
  File "/home/openhabian/.local/lib/python3.13/site-packages/solaredge2mqtt/services/modbus/__init__.py", line 82, in async_init
    await self.read_device_info(
        SunSpecMeterInfoRegister, meter.identifier, meter.offset
    )
  File "/home/openhabian/.local/lib/python3.13/site-packages/solaredge2mqtt/services/modbus/__init__.py", line 121, in read_device_info
    raw_data = await self._read_from_modbus(registers, offset)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/openhabian/.local/lib/python3.13/site-packages/solaredge2mqtt/services/modbus/__init__.py", line 220, in _read_from_modbus
    data = register_or_bundle.decode_response(result.registers, data)
  File "/home/openhabian/.local/lib/python3.13/site-packages/solaredge2mqtt/services/modbus/sunspec.py", line 259, in decode_response
    value = ModbusTcpClient.convert_from_registers(
        registers, self.value_type.data_type, word_order=self.wordorder()
    )
  File "/home/openhabian/.local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 720, in convert_from_registers
    return byte_list.decode(string_encoding)
           ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc2 in position 5: invalid continuation byte

```

puni2k, 2025-05-28:
One question since I have a similar setup here:
Did you configure anything special to read data for the primary inverter?

After adding a follower inverter solaredge2mqtt fails to read any data for the primaries' Modbus ID. Was working fine before the follower inverter was hooked up. First unit arrived with a dead modbus module and was waiting for its replacement.

```
2025-05-28 07:52:21.396 | DEBUG    | solaredge2mqtt.service:__init__:55 - interval=10 logging_level=DEBUG modbus=ModbusSettings(host='192.168.100.132', port=1502, timeout=2, unit=1, check_grid_status=False) mqtt=MQTTSettings(client_id='solaredge2mqtt', broker='192.168.100.16', port=1883, username=None, password=None, topic_prefix='solaredge') location=None prices=PriceSettings(consumption=None, delivery=None, currency=None) monitoring=None wallbox=None influxdb=None weather=None forecast=None homeassistant=None

2025-05-28 07:52:21.396 | INFO     | solaredge2mqtt.services.modbus:__init__:47 - Using SolarEdge inverter via modbus: 192.168.100.132:1502

2025-05-28 07:52:21.396 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: intervalbasetriggerevent

2025-05-28 07:52:21.397 | INFO     | solaredge2mqtt.service:main_loop:115 - Starting SolarEdge2MQTT service...

2025-05-28 07:52:21.397 | INFO     | solaredge2mqtt.service:main_loop:116 - Version: 1.4.1

2025-05-28 07:52:21.399 | INFO     | solaredge2mqtt.service:main_loop:117 - Operationg system: Linux-6.8.12-11-pve-x86_64-with-glibc2.36 (Linux/x86_64)

2025-05-28 07:52:21.399 | DEBUG    | solaredge2mqtt.service:main_loop:120 - interval=10 logging_level=DEBUG modbus=ModbusSettings(host='192.168.100.132', port=1502, timeout=2, unit=1, check_grid_status=False) mqtt=MQTTSettings(client_id='solaredge2mqtt', broker='192.168.100.16', port=1883, username=None, password=None, topic_prefix='solaredge') location=None prices=PriceSettings(consumption=None, delivery=None, currency=None) monitoring=None wallbox=None influxdb=None weather=None forecast=None homeassistant=None

2025-05-28 07:52:21.400 | INFO     | solaredge2mqtt.service:main_loop:121 - Timezone: Europe/Berlin

2025-05-28 07:52:21.400 | INFO     | solaredge2mqtt.core.mqtt:__init__:21 - Using MQTT broker: 192.168.100.16:1883

2025-05-28 07:52:21.400 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: mqttpublishevent

2025-05-28 07:52:21.400 | INFO     | solaredge2mqtt.core.events:subscribe:31 - Event subscribed: mqttsubscribeevent

2025-05-28 07:52:21.403 | INFO     | solaredge2mqtt.services.modbus:async_init:63 - Initializing modbus

2025-05-28 07:52:23.205 | INFO     | solaredge2mqtt.services.modbus:read_device_info:123 - Found inverter SolarEdge SE10K-RWB48BFN4 *redacted*

2025-05-28 07:52:24.159 | INFO     | solaredge2mqtt.services.modbus:read_device_info:123 - Found meter0 SolarEdge SE-MTR-3Y-400V-A *redacted*

2025-05-28 07:52:24.647 | INFO     | solaredge2mqtt.services.modbus:read_device_info:123 - Found battery0 SolarEdge SolarEdge Home Battery 9.2kWh *redacted*

2025-05-28 07:52:31.749 | INFO     | solaredge2mqtt.services.modbus:async_init:105 - Reading modbus registers

No response received after 1 retries, continue with next request

2025-05-28 07:52:35.870 | DEBUG    | solaredge2mqtt.services.modbus:_read_from_modbus:223 - Modbus read exception: Modbus Error: [Input/Output] No response received after 1 retries, continue with next request

2025-05-28 07:52:35.870 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:224 - Unreadable register 40071

2025-05-28 07:52:35.870 | INFO     | solaredge2mqtt.services.modbus:_block_register:231 - Block unreadable register beginning at 40071

No response received after 1 retries, continue with next request

2025-05-28 07:52:39.873 | DEBUG    | solaredge2mqtt.services.modbus:_read_from_modbus:223 - Modbus read exception: Modbus Error: [Input/Output] No response received after 1 retries, continue with next request

2025-05-28 07:52:39.873 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:224 - Unreadable register 40190

2025-05-28 07:52:39.873 | INFO     | solaredge2mqtt.services.modbus:_block_register:231 - Block unreadable register beginning at 40190

No response received after 1 retries, continue with next request

2025-05-28 07:52:43.877 | DEBUG    | solaredge2mqtt.services.modbus:_read_from_modbus:223 - Modbus read exception: Modbus Error: [Input/Output] No response received after 1 retries, continue with next request

2025-05-28 07:52:43.877 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:224 - Unreadable register 57712

2025-05-28 07:52:43.877 | INFO     | solaredge2mqtt.services.modbus:_block_register:231 - Block unreadable register beginning at 57712

2025-05-28 07:52:50.878 | INFO     | solaredge2mqtt.services.modbus:async_init:112 - Modbus initialized

2025-05-28 07:52:50.879 | WARNING  | solaredge2mqtt.services.modbus:async_init:114 - Not readable registers: {57712, 40190, 40071}

2025-05-28 07:52:50.981 | DEBUG    | solaredge2mqtt.services.modbus:_map_inverter:235 - Inverter raw:

2025-05-28 07:52:50.981 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop

2025-05-28 07:53:00.994 | DEBUG    | solaredge2mqtt.services.modbus:_map_inverter:235 - Inverter raw:

2025-05-28 07:53:00.994 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop

[... cut ...]
2025-05-28 07:53:11.007 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop

2025-05-28 07:53:21.023 | DEBUG    | solaredge2mqtt.services.modbus:_map_inverter:235 - Inverter raw:

2025-05-28 07:53:21.023 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid modbus data, skipping this loop
```
Had to switch to a fork of mbmd which adds battery support. But mbmd is also missing json support and some registers of interest to me (i.e. status). Also the documentation is a nightmare.

mihadoo, 2025-05-28:
For the primary inverter that works OK I have (only the important lines in .venv):

SE2MQTT_INTERVAL=60
SE2MQTT_MODBUS__HOST=192.168.218.21
SE2MQTT_MODBUS__PORT=9011
SE2MQTT_MODBUS__TIMEOUT=5
SE2MQTT_MODBUS__UNIT=1
SE2MQTT_MQTT__TOPIC_PREFIX=bridge/t15/solaredgecellar

For the slave inverter where se2mqtt returns an error I have:

SE2MQTT_INTERVAL=60
SE2MQTT_MODBUS__HOST=192.168.218.21
SE2MQTT_MODBUS__PORT=9011
SE2MQTT_MODBUS__TIMEOUT=5
SE2MQTT_MODBUS__UNIT=2
SE2MQTT_MQTT__TOPIC_PREFIX=bridge/t15/solaredgegarage

puni2k, 2025-05-28:
Thanks for you reply.
Don't think that my issue is due to multiple TCP connections. If it were I would also have issues with the initial query of the device info, but that never fails to work.
The only other client that could communicate with the primary inverter would be mbmd and i always stop its container before doing my tests with solaredge2mqtt.

But thanks for the tip with modbus-proxy. I'm sure that will come in handy when our issues might get resolved.

puni2k, 2025-05-31:
More or less. They should have been put into service at the same time, but since the slave arrived broken I had just the master running for 4 weeks until a spare arrived.

Until then solaredge2mqtt worked flawlessly. 
All a bit strange if you ask me since mbmd works fine and can reliably get data from the master inverter, its meter / battery and the slave inverter.

DerOetzi, 2025-05-31:
@puni2k with the help of @mihadoo I'm extending the service to handle master - slave setups in one instance. Will take some days, if succesfully test it, I will inform you so you can test as well.

closed 2025-07-27 by DerOetzi