DerOetzi/solaredge2mqtt #157: no more messages

recumbentseeker, 2025-06-09:
I just started to use solaredge2mqtt it already worked today but now no more message on mqtt anymore:

```
6:06 raspberrypi solarscript.sh[166623]: 2025-06-09 14:26:06.196 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - Invalid powerflow data, skipping this loop
Jun 09 14:26:11 raspberrypi solarscript.sh[166623]: 2025-06-09 14:26:11.525 | INFO     | solaredge2mqtt.services.modbus:_map_inverter:243 - Inverter (SolarEdge SE10K-RWB48BFN4 7B06EEE8): 4, AC 9654.0 W, DC 9801.0 W, 19114.94 kWh, Grid status None,
Jun 09 14:26:11 raspberrypi solarscript.sh[166623]: 2025-06-09 14:26:11.527 | INFO     | solaredge2mqtt.services.modbus:_map_meters:271 - meter0 (SolarEdge MTR-240-3PC1-D-A-MW 6325bbf7): 10328.0 W, 2360.516 kWh, 11854.597 kWh
Jun 09 14:26:11 raspberrypi solarscript.sh[166623]: 2025-06-09 14:26:11.529 | INFO     | solaredge2mqtt.services.modbus:_map_batteries:297 - battery0 (SolarEdge SolarEdge Home Battery 9.2kWh 7B06EEE8): 6, 0.0 W, 98.89 %
Jun 09 14:26:11 raspberrypi solarscript.sh[166623]: 2025-06-09 14:26:11.530 | WARNING  | solaredge2mqtt.services.powerflow.models:is_valid:76 - Consumer used production + grid delivery is not equal to inverter production
Jun 09 14:26:11 raspberrypi solarscript.sh[166623]: 2025-06-09 14:26:11.531 | INFO     | solaredge2mqtt.services.powerflow:calculate_powerflow:77 - powerflow
Jun 09 14:26:11 raspberrypi solarscript.sh[166623]: 2025-06-09 14:26:11.532 | WARNING  | solaredge2mqtt.core.events:_notify_listener:91 - **Invalid powerflow data, skipping this loop**
```

Is "Grid status None" the problem?
Restarting the services did not change anything, status messages I see regularly.

DerOetzi, 2025-06-09:
Do you have any other producers or batteries in your system beside this solaredge inverter or a follower system, as grid power is higher then ac power of the inverter. This is the validation error. Grid status is not the problem this only available if you have equipment like the home backup interface attached

recumbentseeker, 2025-06-09:
Yes I have a separate producer, an other solar panel.

recumbentseeker, 2025-06-10:
Thanks, and thanks for sharing the project.

Are any Modbus messages for setting parameters planned or implemented already?

DerOetzi, 2025-06-11:
Yes but as my own inverter has some problems at the moment i can't finish the control things at the moment. The installer has to come and do some measurements for solaredge to investigate the problem. Therefore i paused the control branch and implement the leader follower things as @mihadoo can test them for me. Will hopefully get some new information because the installer will come tomorrow.

closed 2025-08-02 by DerOetzi