DerOetzi/solaredge2mqtt #181: MQTT Retain flags

gbonny1982, 2025-08-03:
Hi, I recently started using SolarEdge2MQTT. The messages are published on my Mosquitto MQTT-broker.
I would like to have more control on message retainment for both discovery messages and sensor data.

Would it be possible to add 2 environment variables to set retain-flags in SolarEdge2MQTT when messages are published on the MQTT-broker:
```
      - SE2MQTT_MQTT__DISCOVERY_RETAIN=[true|false]
      - SE2MQTT_MQTT__SENSOR_RETAIN=[true|false]
```

(A minor suggestion: I would rename all environment variables containing "HOMEASSISTANT" to "MQTT__DISCOVERY", that would reflect the functionality (and not an APP) as I understand it correctly. Maybe using a migration path so it doesn't break anything: Now it supports the first, then support both, later just the latter.)

DerOetzi, 2025-08-03:
Thanks for your suggestion! I’ll add support for retain flags via these environment variables:

```yaml
- SE2MQTT_HOMEASSISTANT__DISCOVERY_RETAIN=[true|false]
- SE2MQTT_MQTT__SENSOR_RETAIN=[true|false]
```

Regarding the naming: I’ll keep the HOMEASSISTANT prefix, since the AutoDiscovery functionality (including device_class, state_class, and message structure) is defined by the Home Assistant community, even though it’s now also understood by other services.

DerOetzi, 2025-08-04:
@mihadoo Thanks for your input on this issue. This make sense in my opinion but there maybe the need in some use cases. So I think I will make three parameters all by default false:

```
- SE2MQTT_HOMEASSISTANT__DISCOVERY_RETAIN=[true|false]
- SE2MQTT_MQTT__POWERFLOW_RETAIN=[true|false]
- SE2MQTT_MQTT__ENERGY_RETAIN=[true|false]
```

mihadoo, 2025-08-05:
Momentary values with retain make more sense when a timestamp of data collection is included like: "timestamp": "2025-08-05T09:59:24.079884"

gbonny1982, 2025-09-09:
Looks good DerOetzi, a lot of options now, thanks! I have not looked into the code though, was that your request as well?

( Btw, is there some kind of changelog or history file to see what was changed/added/fixed in which version? Just curious :) ) 

> Hi [@gbonny1982](https://github.com/gbonny1982),
> 
> can you check the actual main. I just added the options for every message type to retain.
> 
> https://github.com/DerOetzi/solaredge2mqtt?tab=readme-ov-file#retain-configuration
> 
> Thanks
> 
> DerOetzi