DerOetzi/solaredge2mqtt #182: no data with follower inverter configured

puni2k, 2025-08-04:
First of all thanks a lot for putting in the effort to add support for multi-inverter setups.
Tried it out this morning but unfortunately i can't seem to make it work.

From the logs:
```
2025-08-04 12:58:17.147 | DEBUG    | solaredge2mqtt.service:__init__:58 - interval=10 logging_level=DEBUG external_production=True modbus=ModbusSettings(unit=1, meter=[True, True, True], battery=[True, True], role=leader, host='192.168.100.132', port=1502, timeout=2, check_grid_status=False, advanced_power_controls=disabled, follower=[ModbusUnitSettings(unit=3, meter=[False, False, False], battery=[False, False], role=follower)]) mqtt=MQTTSettings(client_id='solaredge2mqtt', broker='192.168.100.16', port=1883, username=None, password=None, topic_prefix='solaredge') location=None prices=PriceSettings(consumption=None, delivery=None, currency=None) monitoring=None wallbox=None influxdb=None weather=None forecast=None homeassistant=None

2025-08-04 12:58:17.147 | INFO     | solaredge2mqtt.services.modbus:__init__:58 - Using SolarEdge inverter via modbus: 192.168.100.132:1502

2025-08-04 12:58:17.147 | DEBUG    | solaredge2mqtt.services.modbus:__init__:64 - Modbus settings: unit=1 meter=[True, True, True] battery=[True, True] role=leader host='192.168.100.132' port=1502 timeout=2 check_grid_status=False advanced_power_controls=disabled follower=[ModbusUnitSettings(unit=3, meter=[False, False, False], battery=[False, False], role=follower)]

2025-08-04 12:58:17.147 | INFO     | solaredge2mqtt.core.events:subscribe:29 - Event subscribed: modbuswriteevent

2025-08-04 12:58:17.148 | INFO     | solaredge2mqtt.core.events:subscribe:29 - Event subscribed: intervalbasetriggerevent

2025-08-04 12:58:17.148 | INFO     | solaredge2mqtt.service:main_loop:125 - Starting SolarEdge2MQTT service...

2025-08-04 12:58:17.148 | INFO     | solaredge2mqtt.service:main_loop:126 - Version: 2.0.0

2025-08-04 12:58:17.150 | INFO     | solaredge2mqtt.service:main_loop:127 - Operationg system: Linux-6.8.12-13-pve-x86_64-with-glibc2.36 (Linux/x86_64)

2025-08-04 12:58:17.150 | DEBUG    | solaredge2mqtt.service:main_loop:131 - interval=10 logging_level=DEBUG external_production=True modbus=ModbusSettings(unit=1, meter=[True, True, True], battery=[True, True], role=leader, host='192.168.100.132', port=1502, timeout=2, check_grid_status=False, advanced_power_controls=disabled, follower=[ModbusUnitSettings(unit=3, meter=[False, False, False], battery=[False, False], role=follower)]) mqtt=MQTTSettings(client_id='solaredge2mqtt', broker='192.168.100.16', port=1883, username=None, password=None, topic_prefix='solaredge') location=None prices=PriceSettings(consumption=None, delivery=None, currency=None) monitoring=None wallbox=None influxdb=None weather=None forecast=None homeassistant=None

2025-08-04 12:58:17.150 | INFO     | solaredge2mqtt.service:main_loop:132 - Timezone: Europe/Berlin

2025-08-04 12:58:17.150 | INFO     | solaredge2mqtt.core.mqtt:__init__:26 - Using MQTT broker: 192.168.100.16:1883

2025-08-04 12:58:17.150 | INFO     | solaredge2mqtt.core.events:subscribe:29 - Event subscribed: mqttpublishevent

2025-08-04 12:58:17.151 | INFO     | solaredge2mqtt.core.events:subscribe:29 - Event subscribed: mqttsubscribeevent

2025-08-04 12:58:17.163 | INFO     | solaredge2mqtt.services.modbus:async_init:84 - Initializing modbus

2025-08-04 12:58:17.265 | INFO     | solaredge2mqtt.services.modbus:detect_devices:112 - Detecting devices for unit: leader

2025-08-04 12:58:19.311 | INFO     | solaredge2mqtt.services.modbus:read_device_info:170 - Found inverter SolarEdge SE10K-RWB48BFN4 xxxxx

2025-08-04 12:58:19.311 | DEBUG    | solaredge2mqtt.services.modbus:detect_devices:120 - Detected inverter: {'c_id': 'SunS', 'c_did': 1, 'c_length': 65, 'c_manufacturer': 'SolarEdge', 'c_model': 'SE10K-RWB48BFN4', 'c_version': '0004.0023.0036', 'c_serialnumber': 'XXXXX', 'c_deviceaddress': 1, 'c_sunspec_did': 103, 'c_sunspec_length': 50, 'meter0': 203, 'meter1': False, 'meter2': False, 'battery0': 112, 'battery1': 255, 'unit': {'unit': 1, 'key': 'leader', 'role': leader}}

2025-08-04 12:58:20.023 | INFO     | solaredge2mqtt.services.modbus:read_device_info:170 - Found meter0 SolarEdge SE-MTR-3Y-400V-A xxxxx

2025-08-04 12:58:20.540 | INFO     | solaredge2mqtt.services.modbus:read_device_info:170 - Found battery0 SolarEdge SolarEdge Home Battery 9.2kWh xxxxx

2025-08-04 12:58:20.540 | INFO     | solaredge2mqtt.services.modbus:detect_devices:112 - Detecting devices for unit: follower0

2025-08-04 12:58:25.125 | INFO     | solaredge2mqtt.services.modbus:read_device_info:170 - Found inverter SolarEdge SE5K-RW0TEBEN4 xxxxx

2025-08-04 12:58:25.125 | DEBUG    | solaredge2mqtt.services.modbus:detect_devices:120 - Detected inverter: {'c_id': 'SunS', 'c_did': 1, 'c_length': 65, 'c_manufacturer': 'SolarEdge', 'c_model': 'SE5K-RW0TEBEN4', 'c_version': '0004.0023.0027', 'c_serialnumber': 'XXXXX', 'c_deviceaddress': 3, 'c_sunspec_did': 103, 'c_sunspec_length': 50, 'meter0': False, 'meter1': False, 'meter2': 8000, 'battery0': 255, 'battery1': 255, 'unit': {'unit': 3, 'key': 'follower0', 'role': follower}}

2025-08-04 12:58:32.230 | INFO     | solaredge2mqtt.services.modbus:check_readable_registers:185 - Checking modbus registers from unit: leader

2025-08-04 12:58:32.661 | INFO     | solaredge2mqtt.services.modbus:check_readable_registers:185 - Checking modbus registers from unit: follower0

ERROR: request ask for id=1 but got id=3, Skipping.

2025-08-04 12:58:36.766 | DEBUG    | solaredge2mqtt.services.modbus:_read_from_modbus:317 - Modbus read exception: Modbus Error: [Input/Output] No response received after 1 retries, continue with next request

No response received after 1 retries, continue with next request

2025-08-04 12:58:36.766 | ERROR    | solaredge2mqtt.services.modbus:_read_from_modbus:318 - Unreadable register 40071

2025-08-04 12:58:36.766 | INFO     | solaredge2mqtt.services.modbus:_block_register:325 - Block unreadable register beginning at 40071

2025-08-04 12:58:43.769 | WARNING  | solaredge2mqtt.services.modbus:async_init:103 - Not readable registers: {40071}

2025-08-04 12:58:51.120 | DEBUG    | solaredge2mqtt.services.modbus:_map_inverter:331 - Inverter raw:

2025-08-04 12:58:51.120 | WARNING  | solaredge2mqtt.core.events:_notify_listener:79 - Invalid modbus data, skipping this loop

2025-08-04 12:59:01.081 | DEBUG    | solaredge2mqtt.services.modbus:_map_inverter:331 - Inverter raw:

2025-08-04 12:59:01.081 | WARNING  | solaredge2mqtt.core.events:_notify_listener:79 - Invalid modbus data, skipping this loop
```
So it recognizes all devices correctly (yeah) but can't read any data.
If I don't configure the follower inverter everything seems to work fine and I can get data for the primary inverter, battery and meter.

puni2k, 2025-08-05:
sure
```
2025-08-05 06:19:22.504 | DEBUG    | solaredge2mqtt.service:__init__:58 - interval=10 logging_level=DEBUG external_production=True modbus=ModbusSettings(unit=3, meter=[True, True, True], battery=[True, True], role=leader, host='192.168.100.132', port=1502, timeout=2, check_grid_status=False, advanced_power_controls=disabled, follower=[]) mqtt=MQTTSettings(client_id='solaredge2mqtt', broker='192.168.100.16', port=1883, username=None, password=None, topic_prefix='solaredge') location=None prices=PriceSettings(consumption=None, delivery=None, currency=None) monitoring=None wallbox=None influxdb=None weather=None forecast=None homeassistant=None
2025-08-05 06:19:22.504 | INFO     | solaredge2mqtt.services.modbus:__init__:58 - Using SolarEdge inverter via modbus: 192.168.100.132:1502
2025-08-05 06:19:22.504 | DEBUG    | solaredge2mqtt.services.modbus:__init__:64 - Modbus settings: unit=3 meter=[True, True, True] battery=[True, True] role=leader host='192.168.100.132' port=1502 timeout=2 check_grid_status=False advanced_power_controls=disabled follower=[]
2025-08-05 06:19:22.504 | INFO     | solaredge2mqtt.core.events:subscribe:29 - Event subscribed: modbuswriteevent
2025-08-05 06:19:22.504 | INFO     | solaredge2mqtt.core.events:subscribe:29 - Event subscribed: intervalbasetriggerevent
2025-08-05 06:19:22.505 | INFO     | solaredge2mqtt.service:main_loop:125 - Starting SolarEdge2MQTT service...
2025-08-05 06:19:22.505 | INFO     | solaredge2mqtt.service:main_loop:126 - Version: 2.0.0
2025-08-05 06:19:22.507 | INFO     | solaredge2mqtt.service:main_loop:127 - Operationg system: Linux-6.8.12-13-pve-x86_64-with-glibc2.36 (Linux/x86_64)
2025-08-05 06:19:22.507 | DEBUG    | solaredge2mqtt.service:main_loop:131 - interval=10 logging_level=DEBUG external_production=True modbus=ModbusSettings(unit=3, meter=[True, True, True], battery=[True, True], role=leader, host='192.168.100.132', port=1502, timeout=2, check_grid_status=False, advanced_power_controls=disabled, follower=[]) mqtt=MQTTSettings(client_id='solaredge2mqtt', broker='192.168.100.16', port=1883, username=None, password=None, topic_prefix='solaredge') location=None prices=PriceSettings(consumption=None, delivery=None, currency=None) monitoring=None wallbox=None influxdb=None weather=None forecast=None homeassistant=None
2025-08-05 06:19:22.507 | INFO     | solaredge2mqtt.service:main_loop:132 - Timezone: Europe/Berlin
2025-08-05 06:19:22.507 | INFO     | solaredge2mqtt.core.mqtt:__init__:26 - Using MQTT broker: 192.168.100.16:1883
2025-08-05 06:19:22.507 | INFO     | solaredge2mqtt.core.events:subscribe:29 - Event subscribed: mqttpublishevent
2025-08-05 06:19:22.507 | INFO     | solaredge2mqtt.core.events:subscribe:29 - Event subscribed: mqttsubscribeevent
2025-08-05 06:19:22.521 | INFO     | solaredge2mqtt.services.modbus:async_init:84 - Initializing modbus
2025-08-05 06:19:22.622 | INFO     | solaredge2mqtt.services.modbus:detect_devices:112 - Detecting devices for unit: leader
2025-08-05 06:19:27.282 | INFO     | solaredge2mqtt.services.modbus:read_device_info:170 - Found inverter SolarEdge SE5K-RW0TEBEN4 xxxxx
2025-08-05 06:19:27.282 | DEBUG    | solaredge2mqtt.services.modbus:detect_devices:120 - Detected inverter: {'c_id': 'SunS', 'c_did': 1, 'c_length': 65, 'c_manufacturer': 'SolarEdge', 'c_model': 'SE5K-RW0TEBEN4', 'c_version': '0004.0023.0027', 'c_serialnumber': 'xxxx', 'c_deviceaddress': 3, 'c_sunspec_did': 103, 'c_sunspec_length': 50, 'meter0': False, 'meter1': False, 'meter2': 8000, 'battery0': 255, 'battery1': 255}
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/app/solaredge2mqtt/__main__.py", line 4, in <module>
      
)
  File "/app/solaredge2mqtt/service.py", line 44, in run
    loop.run_until_complete(service.main_loop())
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 691, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/app/solaredge2mqtt/service.py", line 147, in main_loop
    await self.powerflow.async_init()
  File "/app/solaredge2mqtt/services/powerflow/__init__.py", line 49, in async_init
    await self.modbus.async_init()
  File "/app/solaredge2mqtt/services/modbus/__init__.py", line 93, in async_init
    await self.detect_devices()
  File "/app/solaredge2mqtt/services/modbus/__init__.py", line 128, in detect_devices
    await self.read_device_info(
  File "/app/solaredge2mqtt/services/modbus/__init__.py", line 158, in read_device_info
    raw_data = await self._read_from_modbus(registers,
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/solaredge2mqtt/services/modbus/__init__.py", line 305, in _read_from_modbus
    data = register_or_bundle.decode_response(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/solaredge2mqtt/services/modbus/sunspec/base.py", line 167, in decode_response
    value = ModbusTcpClient.convert_from_registers(
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/venv/lib/python3.12/site-packages/pymodbus/client/mixin.py", line 719, in convert_from_registers
    return byte_list.decode(string_encoding)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc2 in position 5: invalid continuation byte
[... cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/solaredge2mqtt/services/modbus/sunspec/base.py", line 167, in decode_response
    value = ModbusTcpClient.convert_from_registers(
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/venv/lib/python3.12/site-packages/pymodbus/client/mixin.py", line 719, in convert_from_registers
    return byte_list.decode(string_encoding)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc2 in position 5: invalid continuation byte
```

DerOetzi, 2025-08-05:
Sorry my fault. I have seen your second inverter has a ghost meter 2 attached (This is a known feature/bug don't know yet from se firmware, they seem to reuse the meter registers for some internal, non-public documented feature
) . so please configure unit 3 again as leader and disable meters auto detection for this test:

```
SE2MQTT_MODBUS__METER0=false
SE2MQTT_MODBUS__METER1=false
SE2MQTT_MODBUS__METER2=false
```

puni2k, 2025-08-05:
```
2025-08-05 13:29:02.861 | DEBUG    | solaredge2mqtt.service:__init__:58 - interval=10 logging_level=DEBUG external_production=True modbus=ModbusSettings(unit=3, meter=[False, False, False], battery=[True, True], role=leader, host='192.168.100.132', port=1502, timeout=2, check_grid_status=False, advanced_power_controls=disabled, follower=[]) mqtt=MQTTSettings(client_id='solaredge2mqtt', broker='192.168.100.16', port=1883, username=None, password=None, topic_prefix='solaredge') location=None prices=PriceSettings(consumption=None, delivery=None, currency=None) monitoring=None wallbox=None influxdb=None weather=None forecast=None homeassistant=None
2025-08-05 13:29:02.861 | INFO     | solaredge2mqtt.services.modbus:__init__:58 - Using SolarEdge inverter via modbus: 192.168.100.132:1502
2025-08-05 13:29:02.861 | DEBUG    | solaredge2mqtt.services.modbus:__init__:64 - Modbus settings: unit=3 meter=[False, False, False] battery=[True, True] role=leader host='192.168.100.132' port=1502 timeout=2 check_grid_status=False advanced_power_controls=disabled follower=[]
2025-08-05 13:29:02.862 | INFO     | solaredge2mqtt.core.events:subscribe:29 - Event subscribed: modbuswriteevent
2025-08-05 13:29:02.862 | INFO     | solaredge2mqtt.core.events:subscribe:29 - Event subscribed: intervalbasetriggerevent
2025-08-05 13:29:02.862 | INFO     | solaredge2mqtt.service:main_loop:125 - Starting SolarEdge2MQTT service...
2025-08-05 13:29:02.862 | INFO     | solaredge2mqtt.service:main_loop:126 - Version: 2.0.0
2025-08-05 13:29:02.864 | INFO     | solaredge2mqtt.service:main_loop:127 - Operationg system: Linux-6.8.12-13-pve-x86_64-with-glibc2.36 (Linux/x86_64)
2025-08-05 13:29:02.864 | DEBUG    | solaredge2mqtt.service:main_loop:131 - interval=10 logging_level=DEBUG external_production=True modbus=ModbusSettings(unit=3, meter=[False, False, False], battery=[True, True], role=leader, host='192.168.100.132', port=1502, timeout=2, check_grid_status=False, advanced_power_controls=disabled, follower=[]) mqtt=MQTTSettings(client_id='solaredge2mqtt', broker='192.168.100.16', port=1883, username=None, password=None, topic_prefix='solaredge') location=None prices=PriceSettings(consumption=None, delivery=None, currency=None) monitoring=None wallbox=None influxdb=None weather=None forecast=None homeassistant=None
2025-08-05 13:29:02.865 | INFO     | solaredge2mqtt.service:main_loop:132 - Timezone: Europe/Berlin
2025-08-05 13:29:02.865 | INFO     | solaredge2mqtt.core.mqtt:__init__:26 - Using MQTT broker: 192.168.100.16:1883
2025-08-05 13:29:02.865 | INFO     | solaredge2mqtt.core.events:subscribe:29 - Event subscribed: mqttpublishevent
2025-08-05 13:29:02.865 | INFO     | solaredge2mqtt.core.events:subscribe:29 - Event subscribed: mqttsubscribeevent
2025-08-05 13:29:02.878 | INFO     | solaredge2mqtt.services.modbus:async_init:84 - Initializing modbus
2025-08-05 13:29:02.984 | INFO     | solaredge2mqtt.services.modbus:detect_devices:112 - Detecting devices for unit: leader
2025-08-05 13:29:08.103 | INFO     | solaredge2mqtt.services.modbus:read_device_info:170 - Found inverter SolarEdge SE5K-RW0TEBEN4 xxxx
2025-08-05 13:29:08.103 | DEBUG    | solaredge2mqtt.services.modbus:detect_devices:120 - Detected inverter: {'c_id': 'SunS', 'c_did': 1, 'c_length': 65, 'c_manufacturer': 'SolarEdge', 'c_model': 'SE5K-RW0TEBEN4', 'c_version': '0004.0023.0027', 'c_serialnumber': 'xxxx', 'c_deviceaddress': 3, 'c_sunspec_did': 103, 'c_sunspec_length': 50, 'meter0': False, 'meter1': False, 'meter2': 8000, 'battery0': 255, 'battery1': 255}
2025-08-05 13:29:15.205 | INFO     | solaredge2mqtt.services.modbus:check_readable_registers:185 - Checking modbus registers from unit: leader
2025-08-05 13:29:31.307 | DEBUG    | solaredge2mqtt.services.modbus:_map_inverter:331 - Inverter raw:
{
    "current": 653,
    "current_scale": -2,
    "l1_voltage": 4116,
    "l2_voltage": 4120,
    "l3_voltage": 4114,
    "l1n_voltage": 2380,
    "l2n_voltage": 2373,
    "l3n_voltage": 2379,
    "voltage_scale": -1,
    "power_ac": 15166,
    "power_ac_scale": -1,
    "frequency": 5001,
    "frequency_scale": -2,
    "power_apparent": 15538,
    "power_apparent_scale": -1,
    "power_reactive": -3377,
    "power_reactive_scale": -1,
    "power_factor": -9760,
    "power_factor_scale": -2,
    "energy_total": 1334734,
    "energy_total_scale": 0,
    "current_dc": 20512,
    "current_dc_scale": -4,
    "voltage_dc": 7506,
    "voltage_dc_scale": -1,
    "power_dc": 15397,
    "power_dc_scale": -1,
    "temperature": 4099,
    "temperature_scale": -2,
    "status": 4,
    "l1_current": 217,
    "l2_current": 218,
    "l3_current": 217
}
2025-08-05 13:29:31.309 | DEBUG    | solaredge2mqtt.services.modbus:_map_inverter:338 - modbus: inverter
2025-08-05 13:29:31.309 | INFO     | solaredge2mqtt.services.modbus:_map_inverter:340 - inverter (SolarEdge SE5K-RW0TEBEN4 xxxx): 4, AC 1516.6 W, DC 1539.7 W, 1334.73 kWh, Grid status None, 
2025-08-05 13:29:31.311 | DEBUG    | solaredge2mqtt.services.powerflow:calculate_powerflow:107 - powerflow
2025-08-05 13:29:31.311 | INFO     | solaredge2mqtt.services.powerflow:calculate_powerflow:108 - Powerflow: PV 1539 W, Inverter 1516 W, House 1516 W, Grid 0 W, Battery 0 W, Wallbox 0 W
```

DerOetzi, 2025-08-05:
okay looks good to me now please try again with leader unit 1 and follower unit 3. But disable all autodetection for meters and battery:

```
SE2MQTT_MODBUS__METER0=false
SE2MQTT_MODBUS__METER1=false
SE2MQTT_MODBUS__METER2=false

SE2MQTT_MODBUS__BATTERY0=false
SE2MQTT_MODBUS__BATTERY1=false

SE2MQTT_MODBUS__FOLLOWER0__METER0=false
SE2MQTT_MODBUS__FOLLOWER0__METER1=false
SE2MQTT_MODBUS__FOLLOWER0__METER2=false

SE2MQTT_MODBUS__FOLLOWER0__BATTERY0=false
SE2MQTT_MODBUS__FOLLOWER0__BATTERY1=false
```

If this hopefully works reenable auto detection step by step, if it's crash post the logs again and which one was enabled, please!

puni2k, 2025-08-06:
SUCCESS!
I had a hunch that my problem was caused by a timing issues and decided to play around a bit with the source code during my lunch break and just added two sleeps
```
   async def check_readable_registers(self):
        self.client = AsyncModbusTcpClient(
            host=self.settings.host,
            port=self.settings.port,
            timeout=self.settings.timeout,
            retries=1,
        )

        async with self.client:
            for unit_key, unit_settings in self.settings.units.items():
                logger.info(
                    f"Checking modbus registers from unit: {unit_key}",)
                await self._get_raw_data(unit_key, unit_settings.unit)
# ------- ADDDED THIS ------- 
                await asyncio.sleep(self.settings.timeout + 5)
# ------------------------------------

    async def get_data(
        self,
    ) -> ModbusUnit:
        units: dict[str, ModbusUnit] = {}

        try:
            for unit_key, unit_settings in self.settings.units.items():
                inverter_raw, meters_raw, batteries_raw = await self._get_raw_data(
                    unit_key, unit_settings.unit
                )

                inverter_data = self._map_inverter(unit_key, inverter_raw)
                meters_data = self._map_meters(unit_key, meters_raw)
                batteries_data = self._map_batteries(unit_key, batteries_raw)

                units[unit_key] = ModbusUnit(
                    info=inverter_data.info.unit,
                    inverter=inverter_data,
                    meters=meters_data,
                    batteries=batteries_data
                )
# ------- ADDDED THIS ------- 
                await asyncio.sleep(self.settings.timeout + 5)
# ------------------------------------
```
So by default solaredge2mqtt seems to query my inverters registers and data with a too short interval. Strangely this doesn't happen during device discovery. There solaredge2mqtt by default has a bigger time gap between the two inverter queries.

commit faa98c5e (2025-09-01): Move Modbus Client async context (#184)

This makes the command sequence of check_readable_registers and get_data similar to read_device_info by moving async client context.

Fix #182